serialport/node-serialport #2567: SerialPort.list is not a function

iamisti, 2022-11-27:
### SerialPort Version

10.5.0

### Node Version

14.16

### Electron Version

12.0.4

### Platform

Microsoft Windows NT 10.0.22621.0 x64

### Architecture

x64

### What steps will reproduce the bug?

Just run the specified node version and you will get the issue (i guess)

### What happens?

When running the code ```const board = new Board();``` I get an error.
Interestingly it only happens when i try to run it inside of an electron nodejs server (renderer).

When I run it with node outside of electron, everything works fine.

The error is the following (HardwareDevicesService.ts is my file):
```
board.js:42 Uncaught TypeError: serialport.list is not a function
    at Board.detect (board.js:42)
    at new Board (board.js:291)
    at addBoard (HardwareDevicesService.ts:24)
```

### What should have happened?

run without errors

Ashkikhan, 2022-12-19:
use python instead of node.js. I really have shitten on node.js since its first first release!

miroslawlis, 2023-02-10:
Hi,
not sure if this help but maybe you are importing SerialPort incorrectly (mind curly bracers :) ). 
Have a look at this example [https://github.com/serialport/electron-serialport/blob/6cf4e28ec9df4d45e73d2ceca1e6e131af588956/renderer.js](https://github.com/serialport/electron-serialport/blob/6cf4e28ec9df4d45e73d2ceca1e6e131af588956/renderer.js )

GazHank, 2023-06-22:
Hi @iamisti are you still experiencing this issue?

yangit, 2024-04-19:
@iamisti Are you sure you are calling .list() on a class itself not on an instance of a class?