serialport/node-serialport #2570: do not work for win11 arm64

nishizawa23, 2022-11-30:
## 💥 Proposal
want to work no win11 arm64 version
### What feature you'd like to see

(A clear and concise description of what the proposal is.)

## Motivation

(Please outline the motivation for the proposal. It's interesting knowing what people are working on and also could help community members make suggestions for work-arounds until the feature is built)

## Pitch

(Please explain why this feature should be implemented and how it would be used.)

nlhnt, 2023-02-28:
Well, I am not sure wether this is a complain about missing arm64 support or a protest post that shouts this lib should not support Win11 on arm64?

Hongdeshuai, 2023-03-22:
This packages doesn't works on my Win11 too
OS Name	Microsoft Windows 11 Pro
Version	10.0.22000 Build 22000
The same approach works on Windows 10 but not on Win11

GazHank, 2023-06-07:
Are you able to provide more info on the issue you are facing with Windows 11?  I've been using the package on win11 since the early betas and never experienced any issues or differences compared to windows 10

GazHank, 2023-06-22:
It looks like windows ARM64 will be possible upon upgrade to node js v20, which is due to be the active LTS release on 2023-10-24.

In the meantime people should be able compile from source for AMR64 on windows if they are comfortable using pre-production versions of node js.