serialport/node-serialport #2582: Parser issue

ozturkib, 2022-12-30:
### SerialPort Version

10.5.0

### Node Version

v18.12.1

### Platform

Microsoft Windows NT 10.0.19045.0 x64"Microsoft Windows NT 10.0.19045.0 x64

### Hardware or chipset of serialport

Virtual com port is used.

### What steps will reproduce the bug?

```
var express 		 	 = require('express');
app 				 	 = express();
server 				 	 = require('http').createServer(app);
io 					 	 = require('socket.io')(server);
server.listen(8080);
app.use(express.static('public'));  

const { SerialPort } = require('serialport')
const { ReadlineParser } = require('@serialport/parser-readline')
const port = new SerialPort({ path: 'COM2', baudRate: 9600 })

const parser = port.pipe(new ReadlineParser({ delimiter: '\r\n' }))
parser.on('data', console.log)
//port.on('data', console.log)
```

### What happens?

The code is almost identical from [here](https://serialport.io/docs/api-parser-readline).

Once I insert the parser, it cannot display anything on the side of server console. However, if I run port.on, there is no issue. Hence, I cannot execute properly none of parsers which are listed in [here](https://serialport.io/docs/api-parsers-overview). 

### What should have happened?

No output with the parser scenario.

dhianica, 2023-01-06:
i got same happened to, and no issue when used port.on

stillalivx, 2023-06-14:
Have you fixed it?

GazHank, 2023-06-14:
If you are still facing issues it might be worth double checking that the connected device is emitting '\r\n' at the end of each line.  I've checked both v10 and v11 of serialport and the parser code seems to work correctly for me.

I only managed to accidentally recreate your issue when I tried to communicate with a device that just emits '\n' instead