serialport/node-serialport #2600: Invalid PortInfo typscript declaration

loucass003, 2023-02-15:
### SerialPort Version

10.5.0

### Node Version

v18.10.0

### Platform

Microsoft Windows NT 10.0.22621.0 x64

### Architecture

x64

### Hardware or chipset of serialport

USB-SERIAL CH340

### What steps will reproduce the bug?

```ts
const [first] = await SerialPort.list();
  
console.log(first);
```

### What happens?

that code gives me back

```{
     path: 'COM5',
     manufacturer: 'wch.cn',
     serialNumber: '6&28CF390B&0&5',
     pnpId: 'USB\\VID_1A86&PID_7523\\6&28CF390B&0&5',
     locationId: 'Port_#0005.Hub_#0001',
     friendlyName: 'USB-SERIAL CH340 (COM5)',
     vendorId: '1A86',
     productId: '7523'
}
```

the output is correct but the type deffinition is wrong

### What should have happened?

on the portInfo definition there is only

```ts
export declare interface PortInfo {
    path: string;
    manufacturer: string | undefined;
    serialNumber: string | undefined;
    pnpId: string | undefined;
    locationId: string | undefined;
    productId: string | undefined;
    vendorId: string | undefined;
}
```

the ``friendlyName`` is missing from this definition but is acutally there in the sent object

bluephoton, 2023-02-21:
After hitting F12 to see signature of list() api, I can see that this should work:

import { PortInfo } from "@serialport/bindings-cpp";

reconbot, 2023-06-07:
If we can get it cross platform it's fine by me. Up to you.