serialport/node-serialport #2602: Sr Software Engineer

VietHungPhan, 2023-02-20:
### SerialPort Version

10.8.0

### Node Version

14.17.0

### Electron Version

19

### Platform

Linux seroiuts02097.sero.gic.ericsson.se 3.10.0-1127.19.1.el7.x86_64 #1 SMP Tue Aug 11 19:12:04 EDT 2020 x86_64 x86_64 x86_64 GNU/Linux

### Architecture

x64

### What steps will reproduce the bug?

When I run th command yarn package

### What happens?

zavepih@seroiuts02097[09:41][zavepih/theia-bdgs/electron]$ yarn package
yarn run v1.22.4
$ yarn clean:dist && electron-builder -c.mac.identity=null --publish never
$ rimraf dist
  • electron-builder  version=22.10.5 os=3.10.0-1127.19.1.el7.x86_64
  • loaded configuration  file=/home/zavepih/theia-bdgs/electron/electron-builder.yml
  • writing effective config  file=dist/builder-effective-config.yaml
  • rebuilding native dependencies  dependencies=drivelist@9.2.4, keytar@7.2.0, msgpackr-extract@2.1.2, native-keymap@2.5.0, node-pty@0.11.0-beta17, nsfw@2.2.3, @serialport/bindings-cpp@10.8.0 platform=linux arch=x64
  • install prebuilt binary  name=keytar version=7.2.0 platform=linux arch=x64
  • install prebuilt binary  name=drivelist version=9.2.4 platform=linux arch=x64
  • build native dependency from sources  name=keytar
                                          version=7.2.0
                                          platform=linux
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.6
    prebuild-install info looking for cached prebuild @ /home/zavepih/.npm/_prebuilds/8f6fb9-keytar-v7.2.0-electron-v89-linux-x64.tar.gz
    prebuild-install http request GET https://github.com/atom/node-keytar/releases/download/v7.2.0/keytar-v7.2.0-electron-v89-linux-x64.tar.gz
    prebuild-install http request Proxy setup detected (Host: www-proxy.seli.gic.ericsson.se, Port: 8080, Authentication: No) Tunneling with httpsOverHttp
    prebuild-install http 404 https://github.com/atom/node-keytar/releases/download/v7.2.0/keytar-v7.2.0-electron-v89-linux-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=15.5.7 runtime=electron arch=x64 libc= platform=linux)
    
  • build native dependency from sources  name=drivelist
                                          version=9.2.4
                                          platform=linux
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.6
    prebuild-install info looking for cached prebuild @ /home/zavepih/.npm/_prebuilds/a296b2-drivelist-v9.2.4-electron-v89-linux-x64.tar.gz
    prebuild-install http request GET https://github.com/balena-io-modules/drivelist/releases/download/v9.2.4/drivelist-v9.2.4-electron-v89-linux-x64.tar.gz
    prebuild-install http request Proxy setup detected (Host: www-proxy.seli.gic.ericsson.se, Port: 8080, Authentication: No) Tunneling with httpsOverHttp
    prebuild-install http 404 https://github.com/balena-io-modules/drivelist/releases/download/v9.2.4/drivelist-v9.2.4-electron-v89-linux-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=15.5.7 runtime=electron arch=x64 libc= platform=linux)
    
  • rebuilding native dependency  name=keytar version=7.2.0
  • rebuilding native dependency  name=drivelist version=9.2.4
  • rebuilding native dependency  name=msgpackr-extract version=2.1.2
  • rebuilding native dependency  name=native-keymap version=2.5.0
  • rebuilding native dependency  name=node-pty version=0.11.0-beta17
  • rebuilding native dependency  name=nsfw version=2.2.3
  • rebuilding native dependency  name=@serialport/bindings-cpp version=10.8.0
  • packaging       platform=linux arch=x64 electron=15.5.7 appOutDir=dist/linux-unpacked
  • copying Electron  source=/home/zavepih/theia-bdgs/node_modules/electron/dist destination=/home/zavepih/theia-bdgs/electron/dist/linux-unpacked
  • asar usage is disabled — this is strongly not recommended  solution=enable asar and use asarUnpack to unpack files that must be externally available
[... 30 lines cut ...]
                                          version=9.2.4
                                          platform=linux
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.6
    prebuild-install info looking for cached prebuild @ /home/zavepih/.npm/_prebuilds/a296b2-drivelist-v9.2.4-electron-v89-linux-x64.tar.gz
    prebuild-install http request GET https://github.com/balena-io-modules/drivelist/releases/download/v9.2.4/drivelist-v9.2.4-electron-v89-linux-x64.tar.gz
    prebuild-install http request Proxy setup detected (Host: www-proxy.seli.gic.ericsson.se, Port: 8080, Authentication: No) Tunneling with httpsOverHttp
    prebuild-install http 404 https://github.com/balena-io-modules/drivelist/releases/download/v9.2.4/drivelist-v9.2.4-electron-v89-linux-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=15.5.7 runtime=electron arch=x64 libc= platform=linux)
    
  • rebuilding native dependency  name=keytar version=7.2.0
  • rebuilding native dependency  name=drivelist version=9.2.4
  • rebuilding native dependency  name=msgpackr-extract version=2.1.2
  • rebuilding native dependency  name=native-keymap version=2.5.0
  • rebuilding native dependency  name=node-pty version=0.11.0-beta17
  • rebuilding native dependency  name=nsfw version=2.2.3
  • rebuilding native dependency  name=@serialport/bindings-cpp version=10.8.0
  • packaging       platform=linux arch=x64 electron=15.5.7 appOutDir=dist/linux-unpacked
  • copying Electron  source=/home/zavepih/theia-bdgs/node_modules/electron/dist destination=/home/zavepih/theia-bdgs/electron/dist/linux-unpacked
  • asar usage is disabled — this is strongly not recommended  solution=enable asar and use asarUnpack to unpack files that must be externally available
  • asar usage is disabled — this is strongly not recommended  solution=enable asar and use asarUnpack to unpack files that must be externally available
  • building        target=AppImage arch=x64 file=dist/teal-1.8.0.AppImage
  • building        target=rpm arch=x64 file=dist/teal-1.8.0.x86_64.rpm
Done in 504.82s.

GazHank, 2023-06-07:
Hi @VietHungPhan would you be able to confirm if you are still facing issues with a [supported version of Electron](https://www.electronjs.org/docs/latest/tutorial/electron-timelines)?

GazHank, 2023-06-08:
thansk @VietHungPhan

closed 2023-06-08 by GazHank

VietHungPhan, 2023-06-09:
Hi Gareth,
I will need your support for the issue below.
NODE_MODULE_VERSION 46. This version of Node.js requires
 
NODE_MODULE_VERSION 98. Please try re-compiling or re-installing
Which nodejs version is good for electron 15.5.7?
I am using nodejs/16.15.0 and electron 15.5.7 to build my application with success. Unfortunately when I run my application, I got the error below.

***@***.***[15:15][zavepih/theia-bdgs/electron]$ ./dist/linux-unpacked/teal --no-sandbox
 
[2469:0608/151543.301158:ERROR:viz_main_impl.cc(161)] Exiting GPU process due to errors during initialization
 
[2485:0608/151543.363716:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is swiftshader, ANGLE is
 
/home/zavepih/theia-bdgs/electron/dist/linux-unpacked/resources/app/node_modules/bindings/bindings.js:121
 
        throw e;
 
        ^
 
  
 
Error: The module '/home/zavepih/theia-bdgs/electron/dist/linux-unpacked/resources/app/node_modules/drivelist/build/Release/drivelist.node'
 
was compiled against a different Node.js version using
 
NODE_MODULE_VERSION 46. This version of Node.js requires
 
NODE_MODULE_VERSION 98. Please try re-compiling or re-installing
 
the module (for instance, using `npm rebuild` or `npm install`).
 
    at process.func [as dlopen] (node:electron/js2c/asar_bundle:5:1800)
 
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1170:18)
 
    at Object.func [as .node] (node:electron/js2c/asar_bundle:5:1800)
 
    at Module.load (node:internal/modules/cjs/loader:982:32)
 
    at Module._load (node:internal/modules/cjs/loader:823:12)
 
    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)
 
    at Module.require (node:internal/modules/cjs/loader:1006:19)
 
    at require (node:internal/modules/cjs/helpers:93:18)
 
    at bindings (/home/zavepih/theia-bdgs/electron/dist/linux-unpacked/resources/app/node_modules/bindings/bindings.js:112:48)
 
    at Object.<anonymous> (/home/zavepih/theia-bdgs/electron/dist/linux-unpacked/resources/app/node_modules/drivelist/js/index.js:25:27) {
 
  code: 'ERR_DLOPEN_FAILED'
 
}
 
Thanks & Regards,
Viet Hung PhanEricsson

GazHank, 2023-06-09:
Hi @VietHungPhan 

based on the message:

> NODE_MODULE_VERSION 46. This version of Node.js requires NODE_MODULE_VERSION 98. Please try re-compiling or re-installing

It looks like you need to update Electron to version 22+.  Per your experience this version incompatibility may only show up at runtime.

Per the electron documentation (https://www.electronjs.org/docs/latest/tutorial/electron-timelines) if you can update to Electron 22, and node 16.17 you should be on a compatible and supported environment (until October of 2023 at least)

If you take a look at https://nodejs.org/en/download/releases#ref-1 you can see a description of what NODE_MODULE_VERSION means, and within the table also depicts how old (and unsupported) NODE_MODULE_VERSION 46 is.  As such I don't think there would be any value to you trying to use node 4 and a really old version serial port and other items, and would instead be advised to update electron if at all possible

If you do get errors once you have upgraded electron then do feel free to reach out for help, several other people here will have had the fun of upgrading electron so am sure will be happy to help

VietHungPhan, 2023-06-09:
Thanks Gareth.
Viet-Hung PhanEricsson Inc.