serialport/node-serialport #2604: Serial port vid/pid/metadata isn't being picked up on Fedora 36

obra, 2023-02-27:
### SerialPort Version

@serialport/list@10.5

### Node Version

v18.14.2

### Platform

Linux fedora 6.1.13-100.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Feb 22 18:13:06 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

### Architecture

x64

### What steps will reproduce the bug?

npx @serialport/list --format=jsonline

### What happens?

The serial port for my device is coming up with the vid/pid pair for the USB controller.

```
[jesse@fedora ~]$ npx @serialport/list --format=jsonline
{"path":"/dev/ttyACM0","vendorId":"8086","productId":"9ded"}
{"path":"/dev/ttyS0"}
{"path":"/dev/ttyS1"}
{"path":"/dev/ttyS10"}
{"path":"/dev/ttyS11"}
{"path":"/dev/ttyS12"}
{"path":"/dev/ttyS13"}
{"path":"/dev/ttyS14"}
{"path":"/dev/ttyS15"}
{"path":"/dev/ttyS16"}
{"path":"/dev/ttyS17"}
{"path":"/dev/ttyS18"}
{"path":"/dev/ttyS19"}
{"path":"/dev/ttyS2"}
{"path":"/dev/ttyS20"}
{"path":"/dev/ttyS21"}
{"path":"/dev/ttyS22"}
{"path":"/dev/ttyS23"}
{"path":"/dev/ttyS24"}
{"path":"/dev/ttyS25"}
{"path":"/dev/ttyS26"}
{"path":"/dev/ttyS27"}
{"path":"/dev/ttyS28"}
{"path":"/dev/ttyS29"}
{"path":"/dev/ttyS3"}
{"path":"/dev/ttyS30"}
{"path":"/dev/ttyS31"}
{"path":"/dev/ttyS4"}
{"path":"/dev/ttyS5"}
{"path":"/dev/ttyS6"}
{"path":"/dev/ttyS7"}
{"path":"/dev/ttyS8"}
{"path":"/dev/ttyS9"}
[jesse@fedora ~]$ node --version
v18.14.2
[jesse@fedora ~]$ uname -a
Linux fedora 6.1.13-100.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Feb 22 18:13:06 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
[jesse@fedora ~]$ 

```

### What should have happened?

The serial port for my device should have come up with the vid/pid pair for the device, as it does on other platforms (including Fedora 37)

```
[liveuser@localhost-live list]$ npx @serialport/list@10.5 --format=jsonline
{"path":"/dev/ttyACM0","manufacturer":"Keyboardio","serialNumber":"65644B841439","pnpId":"usb-Keyboardio_Model_100_65644B841439-if00","vendorId":"3496","productId":"0006"}
{"path":"/dev/ttyS0"}
{"path":"/dev/ttyS1"}
{"path":"/dev/ttyS10"}
{"path":"/dev/ttyS11"}
{"path":"/dev/ttyS12"}
{"path":"/dev/ttyS13"}
{"path":"/dev/ttyS14"}
{"path":"/dev/ttyS15"}
{"path":"/dev/ttyS16"}
{"path":"/dev/ttyS17"}
{"path":"/dev/ttyS18"}
{"path":"/dev/ttyS19"}
{"path":"/dev/ttyS2"}
{"path":"/dev/ttyS20"}
{"path":"/dev/ttyS21"}
{"path":"/dev/ttyS22"}
{"path":"/dev/ttyS23"}
{"path":"/dev/ttyS24"}
{"path":"/dev/ttyS25"}
{"path":"/dev/ttyS26"}
{"path":"/dev/ttyS27"}
{"path":"/dev/ttyS28"}
{"path":"/dev/ttyS29"}
{"path":"/dev/ttyS3"}
{"path":"/dev/ttyS30"}
{"path":"/dev/ttyS31"}
{"path":"/dev/ttyS4"}
{"path":"/dev/ttyS5"}
{"path":"/dev/ttyS6"}
{"path":"/dev/ttyS7"}
{"path":"/dev/ttyS8"}
{"path":"/dev/ttyS9"}

Linux localhost-live 6.0.7-301.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Nov 4 18:35:48 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

### Additional information

The issue appears to be reliably reproducible on Fedora 36 but not on Fedora 37. 

On the one hand, "it's better on the current release", but I've got users complaining that my tools are broken and I'd love to figure out a fix for them that isn't "upgrade your OS"

How can I most usefully help figure out what's going wrong and how to fix it?

Thanks!

GazHank, 2023-06-27:
Fixed by OS update per linked issue

closed 2023-06-27 by GazHank

obra, 2023-06-28:
There are still a lot of end-users in the field with OS releases that are getting bitten by this bug. (Minimally, I'm seeing the same behavior on Debian Bullseye.)

Is it serialport's policy to only support the latest release of a given OS?

obra, 2023-06-28:
Ok. I've just verified that this issue (on Debian) is fixed by the changeset described in https://github.com/serialport/bindings-cpp/issues/115#issuecomment-1579566190

I suspect that this issue is the same as https://github.com/serialport/bindings-cpp/issues/115