serialport/node-serialport #2612: Can't read data from GPS

Hongdeshuai, 2023-03-22:
### SerialPort Version

10

### Node Version

14.15.1

### Platform

Windows

### Architecture

x64

### Hardware or chipset of serialport

u-blox AG

### What steps will reproduce the bug?

open serialport
there is no data stream coming in
The events invoked are "on-resume", "on-readable"
No events happen after that even "error" or "close"

### What happens?

===============port resume=============== 
{
  ports: [
    {
      path: 'COM9',
      manufacturer: 'u-blox AG',
      serialNumber: undefined,
      pnpId: 'ROOT\\UBLOXVCP\\0000',
      locationId: undefined,
      friendlyName: 'u-blox Virtual COM Port (COM9)',
      vendorId: undefined,
      productId: undefined
    }
  ]
}
===============port readable=============== 

This is the output, even I tried to log all the port events, on-resume and on-readable invoked

### What should have happened?

Need to read data from GPS device (u-blox AG)
I can read data using Putty and u-center

### Additional information

This is code that I use
`import { SerialPort, ReadlineParser } from "serialport";

// const serialport = new SerialPort({ path: 'COM9', baudRate: 9600 })
SerialPort.list().then((ports) => {
  console.log({ ports });
});

const port = new SerialPort({ path: "COM9", baudRate: 9600 });
const parser = port.pipe(new ReadlineParser());
parser.on("data", (data) => console.log({ data }));
parser.on("close", () => {
  console.log("===============port closing===============");
});

parser.on("readable", (data) => {
  console.log("===============port readable===============", data);
});
parser.on("resume", (data) => {
  console.log("===============port resume===============", data);
});

// ROBOT ONLINE

// Creating the parser and piping can be shortened to
// const parser = port.pipe(new ReadlineParser())
process.stdin.resume(); // keep script running
`

Hongdeshuai, 2023-03-22:
As I confirmed, the same code works on Windows10 but not on Windows11

GazHank, 2023-06-06:
Are you able to confirm which steps of the code appear to work correctly on Windows 10, but incorrectly on windows 11?  I've not seen any issue with compatability from even the very early days of win 11.