serialport/node-serialport #2614: text background extended with auto-scroll

tdjastrzebski, 2023-03-26:
### SerialPort Version

10.4.0

### Node Version

16.15.1

### Hardware or chipset of serialport

VSCode bash terminal + ST Link V3 serial port

### What steps will reproduce the bug?

```C
#define B_RED_BG(msg) "\e[1;101;30m" msg "\e[0m"
printf(B_RED_BG("line 1") "\n");
printf(B_RED_BG("line 2\n"));
printf(B_RED_BG("line 3") "\n");
printf(B_RED_BG("line 4") "\n");
```

### What happens?

If on previous line terminal settings were reset to default (`\e[0m`) after line terminator (`\n`), background of the **next** line gets extended to the end of the line.
[image]

### What should have happened?

I guess, whether or not `\n` line terminator precedes the `\e[0m` reset sequence should not matter, but if it should, then the current line should be affected, not the next one.

reconbot, 2023-03-26:
This is beyond serialport's purview

hellerim, 2023-03-26:
Hi Thomasz,

 

 

The output looks exactly like what I would expect.  What makes you believe that an escape sequence issued after a newline affects the previous line?

First, I think order matters. Second, if so, then the terminating escape has no effect since it is immediately followed be the opening escape sequence of the following printf() statement.

Try 

 
#define B_RED_BG(msg) "\e[1;101;30m" msg "\e[0m"
printf(B_RED_BG("line 1") "\n");
printf("\e[1;101;30m“ „line2\n“);
printf(B_RED_BG("line 3") "\n");
printf(B_RED_BG("line 4") "\n");

 

The output should be the same as before.

 

As a second variant, try this:

 

#define B_RED_BG(msg) "\e[1;101;30m" msg "\e[0m"
printf(B_RED_BG("line 1") "\n");
printf(B_RED_BG("line 3") "\n");
printf(B_RED_BG("line 4") "\n");
printf(B_RED_BG("line 2\n"));

 

The last printed line (with “line 2”) should be completely red, and then be follow by an empty black line.

 

Kind regards,

 

Goetz

 

 

 

Von: T

 

omasz Jastrzębski ***@***.***> 
Gesendet: Sonntag, 26. März 2023 14:27
An: serialport/node-serialport ***@***.***>
Cc: Subscribed ***@***.***>
Betreff: [serialport/node-serialport] text background extended with auto-scroll (Issue #2614)

 

SerialPort Version

10.4.0

Node Version

16.15.1

Electron Version

No response

Platform

No response

Architecture

No response

Hardware or chipset of serialport

VSCode bash terminal + ST Link V3 serial port

What steps will reproduce the bug?

#define B_RED_BG(msg) "\e[1;101;30m" msg "\e[0m"
printf(B_RED_BG("line 1") "\n");
printf(B_RED_BG("line 2\n"));
printf(B_RED_BG("line 3") "\n");
printf(B_RED_BG("line 4") "\n");

What happens?

If on previous line terminal settings were reset to default (\e[0m) after line terminator (\n), background of the next line gets extended to the end of the line.
 <https://user-images.githubusercontent.com/7325277/227775007-98742925-61e7-4ab1-8e6b-d0b6abdd5f16.png> 

What should have happened?

I guess, whether or not \n line terminator precedes the \e[0m reset sequence should not matter, but if it should, then the current line should be affected, not the next one.

Additional information

No response

hellerim, 2023-03-26:
By the way, this is not node programming, but shell programming. And in your code, the escape sequence is not applied to the input line which contains it but to the output. If you send a newline, then a new line is started. If the newline is followed by an escape sequence, then, of course, this will be applied to the current line which is the new one. What was the current line before sending a newline is not current any more after sending it!

 

Kind regards,

 

Goetz

 

 <https://user-images.githubusercontent.com/7325277/227775007-98742925-61e7-4ab1-8e6b-d0b6abdd5f16.png>

tdjastrzebski, 2023-03-26:
@hellerim I see what happens. If "carriage" moves to the new line while background is set, the entire new line background is immediately painted with the current selected background color, even if background is reset to default immediately after and before any subsequent character is sent.  
I would say, it is a strange behavior, but if it is expected and according to specs then it is not a bug.

tdjastrzebski, 2023-03-26:
I see the same effect also when the reset sequence `\e[0m` is followed by new line '\n' and this new line causes scroll.
This makes results a bit nondeterministic, but that is another issue.

hellerim, 2023-03-26:
HI Thomasz,

 

if this is so, then you have an issue with the bash shell but not with Node.js. Please, check the bash release docs. If you don’t find anything about this behavior, send them a bug report.

 

Kind regards,

 

Goetz

closed 2023-03-27 by tdjastrzebski