serialport/node-serialport #2617: Electron 17 and nodejs 16 issue with serial port

khanziaullah, 2023-04-04:
### SerialPort Version

10

### Node Version

16

### Electron Version

17

### Platform

Linux beaglebone 4.19.94-ti-r73 #1buster SMP PREEMPT Fri Apr 15 21:38:30 UTC 2022 armv7l GNU/Linux

### Architecture

armv7l

### Hardware or chipset of serialport

neo-6mv2

### What steps will reproduce the bug?

Hardware: Beaglebone Black - Rev C (armv7l)
Operating System: Linux LXDE
Code: [https://github.com/serialport/electron-serialport#to-use]
Command: npm run start 

### What happens?

App threw an error during load
Error: No native build was found for platform=linux arch=arm runtime=electron abi=114 uv=1 armv=default libc=glibc node=18.14.0 electron=24.0.0
    loaded from: /usr/local/lib/node_modules/serialport/node_modules/@serialport/bindings-cpp

    at load.resolve.load.path (/usr/local/lib/node_modules/serialport/node_modules/node-gyp-build/node-gyp-build.js:60:9)
    at load (/usr/local/lib/node_modules/serialport/node_modules/node-gyp-build/node-gyp-build.js:22:30)
    at Object.<anonymous> (/usr/local/lib/node_modules/serialport/node_modules/@serialport/bindings-cpp/dist/load-bindings.js:11:46)
    at Module._compile (node:internal/modules/cjs/loader:1241:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1296:10)
    at Module.load (node:internal/modules/cjs/loader:1096:32)
    at Module._load (node:internal/modules/cjs/loader:937:12)
    at f._load (node:electron/js2c/asar_bundle:2:13330)
    at Module.require (node:internal/modules/cjs/loader:1120:19)
    at require (node:internal/modules/cjs/helpers:103:18)
A JavaScript error occurred in the main process
Uncaught Exception:
Error: No native build was found for platform=linux arch=arm runtime=electron abi=114 uv=1 armv=default libc=glibc node=18.14.0 electron=24.0.0
    loaded from: /usr/local/lib/node_modules/serialport/node_modules/@serialport/bindings-cpp

    at load.resolve.load.path (/usr/local/lib/node_modules/serialport/node_modules/node-gyp-build/node-gyp-build.js:60:9)
    at load (/usr/local/lib/node_modules/serialport/node_modules/node-gyp-build/node-gyp-build.js:22:30)
    at Object.<anonymous> (/usr/local/lib/node_modules/serialport/node_modules/@serialport/bindings-cpp/dist/load-bindings.js:11:46)
    at Module._compile (node:internal/modules/cjs/loader:1241:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1296:10)
    at Module.load (node:internal/modules/cjs/loader:1096:32)
    at Module._load (node:internal/modules/cjs/loader:937:12)
    at f._load (node:electron/js2c/asar_bundle:2:13330)
    at Module.require (node:internal/modules/cjs/loader:1120:19)
    at require (node:internal/modules/cjs/helpers:103:18)
[6229:0404/152724.791595:ERROR:gl_surface_egl.cc(321)] No suitable EGL configs found.
[6229:0404/152724.793406:ERROR:gl_context_egl.cc(140)] eglGetConfigAttrib failed with error EGL_BAD_CONFIG
[6229:0404/152724.795750:ERROR:gl_surface_egl.cc(321)] No suitable EGL configs found.
[6229:0404/152724.796593:ERROR:gl_surface_egl.cc(1110)] eglCreatePbufferSurface failed with error EGL_BAD_CONFIG
[6229:0404/152724.797226:ERROR:gpu_info_collector.cc(80)] gl::GLContext::CreateOffscreenGLSurface failed
[6229:0404/152724.797749:ERROR:gpu_info_collector.cc(401)] Could not create surface for info collection.
[6229:0404/152724.798240:ERROR:gpu_init.cc(88)] CollectGraphicsInfo failed.
[6229:0404/152724.849578:ERROR:viz_main_impl.cc(186)] Exiting GPU process due to errors during initialization
[6238:0404/152725.659981:ERROR:gpu_memory_buffer_support_x11.cc(49)] dri3 extension not supported.
/home/debian/source/repos/electron-serialport/node_modules/electron/dist/electron exited with signal SIGINT

### What should have happened?

the nmea packets for the configured devices should appear on the screen.
The matching pure nodejs application prints the nmea packets

GazHank, 2023-06-06:
HI @khanziaullah can you confirm which versions of node and elecron you are using?  the exception info doesnt seem to match the error messages (and if you are using elecctron 17 then an update might be needed as it is no longer supported https://www.electronjs.org/docs/latest/tutorial/electron-timelines )

andres-mestra, 2023-07-27:
Hi @GazHank, I have the same error and I am running the latest version of electron@25.3.1 and electron-forge@6.2.1

GazHank, 2023-07-28:
Thanks @andres-mestra would you be able to confirm your error message(s) and the version serialport that you are using?

khanziaullah, 2023-07-28:
> Hi @GazHank, I have the same error and I am running the latest version of electron@25.3.1 and electron-forge@6.2.1

What application are you trying to build? I want to understand the use case.

andres-mestra, 2023-07-30:
> Thanks @andres-mestra would you be able to confirm your error message(s) and the version serialport that you are using?

I am using version 11, however this happens since version 10
I opened an [issue](https://github.com/electron/forge/issues/3282) on the electron forge repository about this problem and the workaround, it's not really a serialport issue

lqd1434, 2023-12-05:
@GazHank I think this error was caused by **node-gyp-build**, and serialport use it
https://github.com/prebuild/node-gyp-build/issues/53
when I use webpack to bundle my electron app, i will got the error, so I have to external serialport.