serialport/node-serialport #2618: Arduino Mega 2560

FrancescaForbicini, 2023-04-07:
### SerialPort Version

10.5.0

### Node Version

18.15.0

### Hardware or chipset of serialport

Arduino Mega 2560

### What steps will reproduce the bug?

I try to receive a string from the arduino. I can communicato with it, but it doesn't when I'm using Arduino Mega 2560 with SerialPort library.

### What happens?

The Arduino can't communicate with me, due to the fact that I'm using SerialPort, because without it, the Arduino works.

### What should have happened?

The arduino should write the same string that I'm giving to it. Without SerialPort, Arduino responds, while when I'm using the library it doesn't work.

### Additional information

I'm using Arduino Mega 2560

akhockey21, 2023-05-19:
I have the same issue with a differen't arduino board

GazHank, 2023-06-06:
Hi, are you able to confirm the operating system you are using, and confirm that nothing else is blocking access to the port while you are trying to communicate with the device?

I've tried with a simple arduino echo sketch and performing the example commands from the https://serialport.io/ main page, and the echo seems to work successfully for me with a Mega 2560, on windows 11, node 18.16.0 and serialport 11.

WaterWolf5918, 2024-01-16:
I have the same problem.