serialport/node-serialport #2649: RPI4 same VID/PID for all devices

DreiDe, 2023-07-03:
### SerialPort Version

10.4.0

### Node Version

v16.20.1

### Platform

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### Architecture

x64

### What steps will reproduce the bug?

Install `serialport` on a Raspberry Pi OS 32bit or 64bit version. Plug in a USB device. List all available ports via `SerialPort.list()`.

### What happens?

No matter which device is plugged in, `node-serialport` always shows the same PID/VID combination of `1106/3483`. At the same time `lsusb` returns a correct response:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 17ef:602d Lenovo Black Silk Keyboard
Bus 001 Device 010: ID 0461:4d22 Primax Electronics, Ltd USB Optical Mouse
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Your website states:
`NodeJS has dropped prebuilt binaries for NodeJS 10 and above for 32bit linux. As a result it's too difficult to maintain support. However if you build nodejs and serialport yourself it will probably work.`

So instead of the default 32bit Raspi OS version I tried the 64bit one. However I got the same result.

### What should have happened?

The devices that were plugged into the RPI show up with their corresponding unique VID/PID combination like with lsusb.

GazHank, 2023-07-03:
Hi @DreiDe i think the website quote you are referring to is specifically in reference to ia32 (intel 32bit platforms) rather than any other architecture so am not sure that is applicable for you.  

It does however look like node js are providing intel 32bit downloads again so it might be that the documentation needs to be updated (again) anyway

Did you try simply installing and using the prebuilt binaries or are you compiling the library yourself from source?

DreiDe, 2023-07-03:
@GazHank I did not built the binaries myself. I just did ˋnpm install serialportˋ. This worked without any errors. However when using the library I do always get the same VID/PID as mentioned above.

GazHank, 2023-07-03:
Thanks for confirming, I'll dig out a raspberry pi and try to replicate

DreiDe, 2023-07-17:
Did you get a chance to test it @GazHank?

GazHank, 2023-07-18:
Hi @DreiDe I've tried to replicate your issue, and the only way I can cause any issue is once updated to Bullseye when it causes issue https://github.com/serialport/node-serialport/issues/2655 to appear.

I wonder if you are simply seeing a different side effect of that same issue :-(

I'll try to get a new release of published ASAP so that we can check if the workaround for the bullseye issue solves this problem for you too

GazHank, 2023-07-20:
Would you be able to check if you still see the same issue now that bindings-cpp release 11.0.3 is available?

DreiDe, 2023-07-22:
Sure, I will check tomorrow.

DreiDe, 2023-08-10:
@GazHank I was just able to test it. I can confirm, that the newest version works fine on the Pi.

Thanks for the fix 👍