serialport/node-serialport #2654: TypeError: "path" is not defined

raed667, 2023-07-12:
### SerialPort Version

11.0.0

### Node Version

v18.14.2

### Platform

Darwin PAR-M1P-XYZ.local 22.4.0 Darwin Kernel Version 22.4.0: Mon Mar  6 20:59:28 PST 2023; root:xnu-8796.101.5~3/RELEASE_ARM64_T6000 arm64

### Architecture

ARM

### What steps will reproduce the bug?

Running the following code crashes the application 

```js
import { SerialPort } from "serialport";

const port = new SerialPort("/dev/tty.usbmodem21202", {
  baudRate: 115200,
  autoOpen: false,
});
```

### What happens?

```sh 
TypeError: "path" is not defined: undefined
    at new SerialPortStream (/Users/xyz/code/slack-cube/node_modules/@serialport/stream/dist/index.js:50:19)
    at new SerialPort (/Users/xyz/code/slack-cube/node_modules/serialport/dist/serialport.js:13:9)
    at file:///Users/xyz/code/slack-cube/index.js:3:14
    at ModuleJob.run (node:internal/modules/esm/module_job:194:25)
```

### What should have happened?

The code shouldn't crash.

closed 2023-07-12 by raed667

Bernard-Murphy, 2023-07-25:
I am also having this issue. Could you please tell us what you did to fix it?

1292277640, 2024-05-08:
It should be:
const port = new SerialPort({path:"/dev/tty.usbmodem21202", 
  baudRate: 115200,
  autoOpen: false,
});