serialport/node-serialport #2655: Issue under Debian Bullseye - observed on a Raspi 4

iKK001, 2023-07-14:
### SerialPort Version

11.0.0

### Node Version

v18.16.1

### Platform

Raspberry Pi OS 32-bit with Debian Bullseye

### Architecture

ARM

### Hardware or chipset of serialport

Profilic 2303, FTDI as well

### What steps will reproduce the bug?

Doing the following:

```
const openPorts = () => {
    SerialPort.list().then((ports) => {
        ports.forEach((port: PortInfo) => {
            console.log(port)
        })
    })
}
```

### What happens?

I get the following console.log 

(In fact, I get this ALWAYS, no matter what FTDI or Profilic USB I connect. It always shows me the same wrong text log):

```
{
  path: '/dev/ttyUSB0',
  manufacturer: undefined,
  serialNumber: undefined,
  pnpId: undefined,
  locationId: undefined,
  vendorId: '1106',
  productId: '3483'
}
```

### What should have happened?

What I do expect:

```
{
  path: '/dev/ttyUSB0',
  manufacturer: 'Prolific_Technology_Inc.',
  serialNumber: XXXXXXXXXXXXX,
  pnpId: undefined,
  locationId: undefined,
  vendorId: '067B',
  productId: '2303'
}
```

### Additional information

I suspect that `Serialport` does not work with Debian Bullseye !

Any ideas what to do ??

GazHank, 2023-07-14:
Hi @iKK001 

I believe this is related to https://github.com/serialport/bindings-cpp/issues/115 which is the result of a defect introduced in debian which is due to be corrected.  Unfortunately it seems like the fix to Debian seems to be taking some time to apply the fix, so there is a proposed workaround that has been proposed https://github.com/serialport/bindings-cpp/pull/116

I'd really like to make sure the workaround doesn't cause any issues or regressions for Distros that haven't applied the broken Udev rules change, but if the number of users affected continues to grow then perhaps we just need to pull the workaround in as-is.

iKK001, 2023-07-14:
Thank you @GazHank, I will try the workaround in [bindings-cpp#116](https://github.com/serialport/bindings-cpp/pull/116)

Since Debian made such an ugly defect, I definitively think that many USB-hotplug issues arise all over the place while this is not fixed. Therefore I think pulling this workaround as is would at least calm down many people.

iKK001, 2023-07-14:
I saw that you merged the PR inside [bindings-cpp](https://github.com/serialport/bindings-cpp) - thank you very much.

However, shouldn't there also be a new version then in order to make it link automatically with `serial` ?

What I mean:

inside my package.json I can only state 

```
"dependencies": {
    "@types/serialport": "^8.0.2",
    "serialport": "^11.0.0"
}
```

But this still does not package the correct newbies you just merged inside bindings-cpp.

I also tried:

```
"dependencies": {
    "@types/serialport": "^8.0.2",
    "@serialport/bindings-cpp": "^11.0.1",
    "serialport": "^11.0.0"
}
```

But that did not help.

I think that there is some sort of `prescript` inside the `serialport` package that also needs some update from you - or can I do that myself ? If yes, how ?

Thank you for clarifying.

GazHank, 2023-07-14:
I've merged the changes, and drafted a new version, but it's not published yet

@reconbot are you able to review the draft release and publish?

GazHank, 2023-07-20:
With the release 11.0.3 of bindings-cpp this should now be fixed.  Please let me know if you continue to see any issues

iKK001, 2023-07-22:
There is already version 12.0.1 out of bindings-cpp. And I tested with it. 

And it works !!! (i.e. your 4 extra lines together with the comment is now part of the `linux-list.js` file as needed).

Thank you very much !

One more question:

Will this also go into `node-serial` repo ? 

Because right now, I need to add the special bindings-cpp version as an extra dependency to my package.json file, such as...

```
"@serialport/bindings-cpp": "^12.0.1",
"serialport": "^11.0.0",
```

And I would like to be able to only add the serialport.

GazHank, 2023-07-22:
You shouldn't need to force v12 of the bindings as the latest v11.0.3 should fix this issue too. 

For the main repo I'm hoping to release a final v11.x which will make sure the recent bindings fixes are fully applied and it will be the last supporting node 14; before a fresh new v12 release.  Sorry it's taking a little time to publish that release, I'm trying to be careful to not break anything for users who are stuck on node 14 for some reason, so that it will give them as much time as possible to upgrade

iKK001, 2023-07-22:
ah ok - all good. Thank you for the explanations. I can wait ! 

It is always better to make things careful than in a hurry.

Thank you very much for your support. I look forward to the upcoming release.

(You can close this ticket from my part, since I can now work with the v11.0.3 fix)

closed 2023-07-22 by GazHank