serialport/node-serialport #2659: SerialPort on Node v20 incurs high CPU load.

mousseq, 2023-07-21:
### SerialPort Version

8.0.2

### Node Version

20.4.0

### Platform

Linux pi4-1 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### Architecture

Raspberry Pi 4 (ARM)

### Hardware or chipset of serialport

FTDI

### What steps will reproduce the bug?

The enclosed program opens three serial ports and reads from them (and does nothing with the data). The ports receive data at relatively modest rates (1000 bytes 30 times a second).

### What happens?

When built and run with Node v19.8.1, the CPU load is negligible. When built and run with Node v20, the CPU load increases dramatically. One of the four cores runs at 100%. The machine is a Raspberry Pi 4 running the Debian Bullseye version of the OS.

### What should have happened?

I expect that the CPU load will not very significantly between the Node 19.8.1 and Node 20.4 versions.
[sptest.tar.gz](https://github.com/serialport/node-serialport/files/12115504/sptest.tar.gz)

GazHank, 2023-07-21:
Would you be able to check if you experience the same issue with node 20.2?

mousseq, 2023-07-21:
Gareth:

Good call! When build and run with Node 20.2 (a test I was going to run later ...), the load is back to manageable levels. None of the cores are heavily loaded.

Thanks:
Adam

GazHank, 2023-07-21:
thanks @mousseq

In that case I strongly suspect this is related to the node js upgrade of libuv which occurred in node 20.3.  Issue https://github.com/serialport/node-serialport/issues/2656 is another symptom of this.

I'm trying to investigate the root cause and get sufficient info about the problem so that we can get a fix applied to libuv, but it's proving a bit messy/time consuming so far.

uncaught, 2023-10-22:
node 20.8.1 and 21.0.0 are still affected.

Thanks for the hint regarding node 20.2. That version works still very well.

robertsLando, 2024-01-17:
News on this?

mgrouch, 2025-01-16:
I have same issue with node 22.13

If node is started at boot time (via systemd server) and a serial device 
(M5stack atomS3 esp32) is plugged
then node uses 100% of one of the cores on raspberry pi5.