serialport/node-serialport #2677: READ DATA SPLITTED

ReisoAphi, 2023-08-05:
### SerialPort Version

11.0.1

### Node Version

18.13.0

### Platform

Linux SYMMETRIC 6.1.0-10-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.38-1 (2023-07-14) x86_64 GNU/Linux

### Architecture

x64

### What steps will reproduce the bug?

I have this code and the issue is that im getting the read splitter in 9 different buffers, shouldn't it be in the same buffer ?

const { SerialPort } = require("serialport")
const devices = new Array
let indexerCollector = 0

const port = new SerialPort({
  path: '/dev/ttyUSB0',
  baudRate: 4800,
  dataBits: 8,
  stopBits: 1,

})

port.on("data",(data)=>{
  console.log(data)

})
port.open(()=>{
    console.log("PORT OPEN")
    port.write(Buffer.from('020300000002C438','hex'),'hex')
    // port.write(Buffer.from('010300000002C40B','hex'),'hex')
})    
port.on("error",(err)=>{
  console.log(err)
})

### What happens?

PORT OPEN
<Buffer 02>
<Buffer 03>
<Buffer 04>
<Buffer 00>
<Buffer ec>
<Buffer 00>
<Buffer 00>
<Buffer 08>
<Buffer c6>

### What should have happened?

I should be getting the buffer like this.

<Buffer 02 03 04 00 ec 00 00 08 c6>

artapon, 2023-10-25:
I found the problem too. Are you find anyway or solution to fixed this problem. Thank.

reconbot, 2023-10-25:
The data comes in slower than the computer can process it. So you will need a parser to help buffer the data or interpret it.

https://serialport.io/docs/api-parsers-overview

https://serialport.io/docs/api-parser-inter-byte-timeout

https://serialport.io/docs/api-parser-byte-length