serialport/node-serialport #2684: Raspi 4 gets stuck after reading for 24 hours from dev/ttyUSB0

akesser, 2023-08-23:
### SerialPort Version

11.0.1

### Node Version

20.5.1

### Platform

linux aarch64 GNU/Linux

### Architecture

ARM64

### What steps will reproduce the bug?

When I try to read from dev/ttyUSB0 on a Raspberry Pi 4, the howl raspy gets stuck after about 24 hours and I always have to restart the machine by force to be able to communicate with it again. I'm using `SerialPortStream` and pipeline from node:stream/promises to handle the data like in the following example:

```js
const s = new SerialPortStream({
	baudRate: 9600,
	path: "/dev/ttyUSB0",
	binding: autoDetect()
});
await pipeline(
  s,
  async (source) => {
    for await (const values of source) {
      await logDataToDatabase(values);
    }
  }
);
```

### What happens?

The Raspi is no longer responding and needs to be updated, also no data is saved into the database

### What should have happened?

The Raspberry Pi should not get stuck but continue to work normally

### Additional information

I tried to monitor the memory usage of the node process, and the free memory of the raspy, but did not see any problems there, I also checked for memory leaks in the code bud did not find any.

akesser, 2023-09-06:
I tried to close the connection to the port and reconnect to circumvent this behavior as describe in #2685, but the problem is still occurring, not every 24 hours, but on the basis of 2 to 3 days now.

devwatts, 2023-11-04:
Did you resolve this issue?

devwatts, 2023-11-04:
@akesser

akesser, 2023-11-04:
I am running the script in a kubernetes pod and have a job that kills the pod every 24 hours. In this way I can circumvent the problem, but solved... no

devwatts, 2023-11-04:
Does this happen with Node v18 or Node v16 ?

I am stuck with this issue

akesser, 2023-11-04:
With node 20 I could not reconnect as described in #2685 , running with node18 on alpine now. I have not tested node v16

devwatts, 2023-11-04:
I have tested with raspbian bullseye node v18 with a Pi 3, Not facing any issue like this

But same setup with Pi 4 this issue occurs

Will try alpine as well

devwatts, 2023-12-06:
Did not work