serialport/node-serialport #2685: Cannot reconnect to port on linux, OSX is working fine

akesser, 2023-08-23:
### SerialPort Version

11.0.1

### Node Version

20.5.1

### Platform

aarch64 GNU/Linux

### Architecture

ARM64

### What steps will reproduce the bug?

When I try to read data from dev/ttyUSB0 for a long time my Raspberry Pi gets stuck, I opened bug #2684 for this. To circumvent the problem, I tried to close the connection to the port and reconnect. Testing this code on a Mac worked without problems, but on the Raspi I get the following error: 

`[Error: Error Resource temporarily unavailable Cannot lock port]`

To connect and reconnect on a given interval I added the following class:

```js

export class RenewableStream extends Readable {
	_source: SerialPortStream;
	_intervalClose = false;
	constructor(serialOprions: OpenOptions, interval: number, options: ReadableOptions = {}) {

		super(options);

		this._source = new SerialPortStream(serialOprions);

		this._source.on("data", (chunk) => {
			this.push(chunk);
		});

		this._source.on("end", () => {
			if (!this._intervalClose) {
				this.push(null);
			}
			console.log("ENDDDD");
		});

		setInterval(() => {
			this._intervalClose = true;
			this._source.close(async (err) => {
				if (err) {
					console.log("close error: ", err);
				}
				this._source.open((err) => {
					if (err) {
						console.log("Connect error: ",err);
					}
				});
				this._intervalClose = false;
			});
		}, interval);
	}

	_read(): void {
		// empty
	}

	async close(): Promise<void> {
		console.log("startClose");
		return new Promise((resolve, reject) => {
			this._source.close((err) => {
				console.log("Close");
				if (err) {
					reject(err);
				}
				resolve();
			});
		});
	}
}
```

And I use the following code to test the code:

```js
const s = new RenewableStream({
	baudRate: 9600,
	path: "/dev/ttyUSB0",
	binding: autoDetect()
}, 10000); // reconnect every ten seconds

await pipeline(
	s,
	async (source) => {
		for await (const d of source) {
			console.log(d.toString());
		}
	}
);
```

### What happens?

Running the code given above runs without any errors on my Mac, but running it on a Raspberry Pi for generates the error `[Error: Error Resource temporarily unavailable Cannot lock port]` when the code tries to connect after `close()` was called.

### What should have happened?

Like on the Mac, the connection should be reopened and should be usable without errors

GazHank, 2023-08-23:
Are you able to check if you get the same issues on Node V18?  This could be related to the other issues being investigated for node v20 on Linux (https://github.com/serialport/node-serialport/issues/2656)

GazHank, 2023-08-23:
Thanks for confirming @akesser