serialport/node-serialport #2707: Doesn't work in Bun

jakeg, 2023-09-22:
### SerialPort Version

12.0.0

### Node Version

Bun 1.0.2

### Platform

Linux thing 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

### Hardware or chipset of serialport

Pi Pico

### What steps will reproduce the bug?

Trying to get serialport working in Bun instead of Node. Please see the issue on the Bun tracker here:

https://github.com/oven-sh/bun/issues/4622#issuecomment-1730902748

### What happens?

see issue on Bun

### What should have happened?

work same as in NodeJS

Dimava, 2025-05-17:
It now opens but doesn't seem to emit `data` messages

Dimava, 2025-05-17:
`asyncRead` function in `node_modules\@serialport\bindings-cpp\dist\load-bindings.js` doesn't call callback

lucas-vitrus, 2025-10-20:
solved?