serialport/node-serialport #2715: Cannot find module 'serialport' on build with cordova-electron

laux98, 2023-10-10:
### SerialPort Version

12.0.0

### Node Version

18.16.1

### Electron Version

25.2.0

### Platform

Ubuntu 22.04

### Architecture

x64

### What steps will reproduce the bug?

Hello, my problem is when I include the library SerialPort in my cordova-electron projet, build crashe on windows 10 , window 11 and Ubuntu 22.04.

In the file "pathToMyProjetct/cordova/platforms/electron/platform_www/cdv-electron-main.js", I include the line 
```const { SerialPort } = require("serialport");```
to use the library. 

### What happens?

Build don't launch and error is displayed.

```
A JavaScript error occurred in the main process
Uncaught Exception:
Error: Cannot find module 'serialport'
Require stack:
- /pathToMyProject/cordova/platforms/electron/build/linux-unpacked/resources/app.asar/cdv-electron-main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:892:15)
    at Function.n._resolveFilename (electron/js2c/browser_init.js:245:1128)
    at Module._load (internal/modules/cjs/loader.js:737:27)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Module.require (internal/modules/cjs/loader.js:964:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/pathToMyProject/cordova/platforms/electron/build/linux-unpacked/resources/app.asar/cdv-electron-main.js:27:24)
    at Module._compile (internal/modules/cjs/loader.js:1083:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1113:10)
    at Module.load (internal/modules/cjs/loader.js:940:32)

```

### What should have happened?

Build launch and I can use serial port.

### Additional information

I am trying to use the serial port library on cordova-electron.
When i use 
```cordova run electron --nobuild ```, this work perfectly and the application is launch and I can use serialport.
When i want to do the build, I use 
```cordova build android```and there are no errors. 
But when I launch the build , this error is displayed :

```
A JavaScript error occurred in the main process
Uncaught Exception:
Error: Cannot find module 'serialport'
Require stack:
- /pathToMyProject/cordova/platforms/electron/build/linux-unpacked/resources/app.asar/cdv-electron-main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:892:15)
    at Function.n._resolveFilename (electron/js2c/browser_init.js:245:1128)
    at Module._load (internal/modules/cjs/loader.js:737:27)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Module.require (internal/modules/cjs/loader.js:964:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/pathToMyProject/cordova/platforms/electron/build/linux-unpacked/resources/app.asar/cdv-electron-main.js:27:24)
    at Module._compile (internal/modules/cjs/loader.js:1083:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1113:10)
    at Module.load (internal/modules/cjs/loader.js:940:32)

```

The line "cdv-electron-main.js:27:24" is 
```
const { SerialPort } = require("serialport");
```

I have the same issue in Windows 10, Windows 11 and Ubuntu22.04.

kikino1989, 2024-08-14:
@laux98 did you find a solution for this? 🙏

twtylkmrh, 2025-01-04:
https://stackoverflow.com/questions/79163239/electron-forge-serialport-native-dependency-packaging-not-working

may be helpful