serialport/node-serialport #2720: Build Error on Different Platforms (serialport & electron)

dkarski, 2023-10-20:
### SerialPort Version

10.5.0

### Node Version

18.16.1

### Electron Version

26.4.0

### Platform

Darwin MacBook-Pro-Appnroll-3.local 22.4.0 Darwin Kernel Version 22.4.0: Mon Mar  6 21:00:17 PST 2023; root:xnu-8796.101.5~3/RELEASE_X86_64 x86_64

### Architecture

x86/x64

### What steps will reproduce the bug?

1. Clone the [dkarski:electron-serialport:electron-builder](https://github.com/dkarski/electron-serialport/tree/electron-builder) project repository.
2. Run the application on a Mac system.
3. Try to build and run the application on a Windows system.

### What happens?

When I try to run the application on a Windows system that has been built on a Mac system, the application behaves incorrectly and displays the following error in the console. However, when I build the application on Windows, it runs without any issues.

```
Uncaught Error: \\?\C:\Users\Daniel Karski\AppData\Local\Programs\electron-serialport\resources\app.asar.unpacked\node_modules\@serialport\bindings-cpp\build\Release\bindings.node is not a valid Win32 application.
\\?\C:\Users\Daniel Karski\AppData\Local\Programs\electron-serialport\resources\app.asar.unpacked\node_modules\@serialport\bindings-cpp\build\Release\bindings.node
```

### What should have happened?

I would expect the application to work correctly on a Windows system, regardless of the system on which it was built.

### Additional information

In my case, I used the electron-builder tool for building the application, but a similar issue occurs when using other tools such as electron-forge with webpack or Vite. It appears that the build tool is not the primary cause of the issue described.

It's important to note that when I build the application on Windows, it runs without any issues. The problem seems to be specific to building the application on a Mac and then running it on Windows.

I would appreciate any assistance in resolving this problem, as it hinders the development and testing of the application on different platforms.

Thank you for your attention and assistance!

GazHank, 2023-10-20:
I'm not sure if electron-builder supports cross build of a windows app from MacOS.  The only cross build scenario that I can see in the documentation is building on Linux to target windows https://www.electron.build/multi-platform-build.html#to-build-app-for-windows-on-linux

I wonder if a docker container would let you work around this limitation?

emily-ntc, 2025-01-24:
facing the same issue. Did you ever figure out a workaround? Did you try the cocker container fix mentioned above?