serialport/node-serialport #2725: NodeGyp Issue on OSX

ScreamZ, 2023-10-26:
### SerialPort Version

6.0.5

### Node Version

18.15

### Platform

Darwin screamz-2.home 23.0.0 Darwin Kernel Version 23.0.0: Fri Sep 15 14:41:43 PDT 2023; root:xnu-10002.1.13~1/RELEASE_ARM64_T6000 arm64

### Architecture

ARM

### What steps will reproduce the bug?

Install any package that ship serial-port like johnny-five

### What happens?

```
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Progress: resolved 1695, reused 1454, downloaded 194, added 273, done
node_modules/.pnpm/@scarf+scarf@1.3.0/node_modules/@scarf/scarf: Running postinstall script, done in 3s
node_modules/.pnpm/serialport@6.0.5/node_modules/serialport: Running install script, failed in 2s
.../node_modules/serialport install$ prebuild-install || node-gyp rebuild
│ prebuild-install WARN install No prebuilt binaries found (target=18.15.0 runtime=node arch=arm64 platform=darwin)
│ gyp info it worked if it ends with ok
│ gyp info using node-gyp@9.4.0
│ gyp info using node@18.15.0 | darwin | arm64
│ gyp info find Python using Python version 3.11.6 found at "/opt/homebrew/opt/python@3.11/bin/python3.11"
│ gyp info spawn /opt/homebrew/opt/python@3.11/bin/python3.11
│ gyp info spawn args [
│ gyp info spawn args   '/Users/screamz/.volta/tools/image/packages/pnpm/lib/node_modules/pnpm/dist/node_modules/node-gyp/gyp/gyp_main.py',
│ gyp info spawn args   'binding.gyp',
│ gyp info spawn args   '-f',
│ gyp info spawn args   'make',
│ gyp info spawn args   '-I',
│ gyp info spawn args   '/Users/screamz/dev-workspace/pro/livecorp/freakshow/node_modules/.pnpm/serialport@6.0.5/node_modules/serialport/build/config.gypi',
│ gyp info spawn args   '-I',
│ gyp info spawn args   '/Users/screamz/.volta/tools/image/packages/pnpm/lib/node_modules/pnpm/dist/node_modules/node-gyp/addon.gypi',
│ gyp info spawn args   '-I',
│ gyp info spawn args   '/Users/screamz/Library/Caches/node-gyp/18.15.0/include/node/common.gypi',
│ gyp info spawn args   '-Dlibrary=shared_library',
│ gyp info spawn args   '-Dvisibility=default',
│ gyp info spawn args   '-Dnode_root_dir=/Users/screamz/Library/Caches/node-gyp/18.15.0',
│ gyp info spawn args   '-Dnode_gyp_dir=/Users/screamz/.volta/tools/image/packages/pnpm/lib/node_modules/pnpm/dist/node_modules/node-gyp',
│ gyp info spawn args   '-Dnode_lib_file=/Users/screamz/Library/Caches/node-gyp/18.15.0/<(target_arch)/node.lib',
│ gyp info spawn args   '-Dmodule_root_dir=/Users/screamz/dev-workspace/pro/livecorp/freakshow/node_modules/.pnpm/serialport@6.0.5/node_modules/serialport',
│ gyp info spawn args   '-Dnode_engine=v8',
│ gyp info spawn args   '--depth=.',
│ gyp info spawn args   '--no-parallel',
│ gyp info spawn args   '--generator-output',
│ gyp info spawn args   'build',
│ gyp info spawn args   '-Goutput_dir=.'
│ gyp info spawn args ]
│ gyp info spawn make
│ gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
│   CXX(target) Release/obj.target/serialport/src/serialport.o
│ ../src/serialport.cpp:41:48: error: too few arguments to function call, single argument 'context' was not specified
│   v8::String::Utf8Value path(info[0]->ToString());
│                              ~~~~~~~~~~~~~~~~~ ^
│ /Users/screamz/Library/Caches/node-gyp/18.15.0/include/node/v8-value.h:360:44: note: 'ToString' declared here
│   V8_WARN_UNUSED_RESULT MaybeLocal<String> ToString(
│                                            ^
│ ../src/serialport.cpp:41:25: error: no matching constructor for initialization of 'v8::String::Utf8Value'
│   v8::String::Utf8Value path(info[0]->ToString());
│                         ^    ~~~~~~~~~~~~~~~~~~~
[... 164 lines cut ...]
│                                    ^
│ ../src/serialport.cpp:413:43: error: expected ';' after top level declarator
│   void init(v8::Handle<v8::Object> target) {
│                                           ^
│                                           ;
│ 7 warnings and 16 errors generated.
│ make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
│ gyp ERR! build error 
│ gyp ERR! stack Error: `make` failed with exit code: 2
│ gyp ERR! stack     at ChildProcess.onExit (/Users/screamz/.volta/tools/image/packages/pnpm/lib/node_modules/pnpm/dist/node_modules/node-gyp/lib/build.js:203:23)
│ gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
│ gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:291:12)
│ gyp ERR! System Darwin 23.0.0
│ gyp ERR! command "/Users/screamz/.volta/tools/image/node/18.15.0/bin/node" "/Users/screamz/.volta/tools/image/packages/pnpm/lib/node_modules/pnpm/dist/node_modules/node-gyp/bin/node-gy…
│ gyp ERR! cwd /Users/screamz/dev-workspace/pro/livecorp/freakshow/node_modules/.pnpm/serialport@6.0.5/node_modules/serialport
│ gyp ERR! node -v v18.15.0
│ gyp ERR! node-gyp -v v9.4.0
│ gyp ERR! not ok 
└─ Failed in 2s at /Users/screamz/dev-workspace/pro/livecorp/freakshow/node_modules/.pnpm/serialport@6.0.5/node_modules/serialport
 ELIFECYCLE  Command failed with exit code 1.   
 ```

### What should have happened?

It should have been installed

GazHank, 2023-10-28:
It looks like you are trying to run a very old version of serialport that is not compatible with Node 18.  I'd encourage you to update to the latest version if you are able.  You can find instructions to help with the upgrade process at https://serialport.io/docs/guide-upgrade

ScreamZ, 2023-10-28:
This is related to this library https://github.com/ajfisher/node-pixel, thanks for you answer

closed 2023-10-28 by ScreamZ