serialport/node-serialport #2726: Invalid serial number in PortInfos (windows)

romainmnr, 2023-11-03:
### SerialPort Version

>=10.5.x

### Node Version

18.x.x

### Platform

Microsoft Windows NT 10.0.19045.0 x64

### Architecture

x64

### What steps will reproduce the bug?

Listing devices from `Serialport.list()` on windows.

### What happens?

Since `serialPort@10.5` the return serial number in the port infos is no longer correct.

Result of `SerialPort.list()`
`pnpId`: `USB\\VID_6244&PID_0622&MI_00\\6&FED21C1&0&0000`
`serialNumber`: `6&FED21C1&0&0000` // wrong SN

### What should have happened?

Expected result of `SerialPort.list()`

`pnpId`: `USB\\VID_6244&PID_0622&MI_00\\6&FED21C1&0&0000`
`serialNumber`: `000002003006`

### Additional information

Issur not detected on macOs

romainmnr, 2023-11-03:
I've also noticed that windows created a new folder in the registry with `&MI_00` at the end.

[image]

drostea, 2024-01-15:
I can confirm that starting with v10.5.0 the Serial Numbers on Windows are not reported correctly.
I also have a device with multiple CDCs that opens up multiple Serial Ports.

Instead of the correct Serial Number the last part of the pnpId is used.

What is also broken in v10.4.0 is that if connect two USB devices with the same reported Serial Number, Windows replaces the SN of the second attached device with a fake Serial Number. I did not found a way with the registry / wmi to query for the correct serial number in this case.