serialport/node-serialport #2748: Cannot work in a worker thread... please save me

61go, 2024-02-27:
### SerialPort Version

12+

### Node Version

v20

### Architecture

64

### What steps will reproduce the bug?

`

// main.ts

import { Worker } from 'worker_threads';

if (require.main === module) {
  const worker = new Worker('./dist/serialWorker.js');

  worker.on('message', (message) => {
    console.log('Received message from worker:', message);
  });

  worker.postMessage({ type: 'START_SERIAL_PORT', port: '/dev/ttyUSB0', baudRate: 9600 });
}`

`// serialWorker.ts

import { parentPort } from 'worker_threads';
import { SerialPort, SerialPortOpenOptions } from 'serialport';

if (parentPort) {
  parentPort.on('message', (message) => {
    if (message.type === 'START_SERIAL_PORT') {
    //   const { port, baudRate } = message;

      const portOptions: SerialPortOpenOptions<any> = {
        path: 'COM4',
        baudRate: 921600,
        dataBits: 8,
        parity: 'none',
        stopBits: 1,
        autoOpen: false,
      };
    //   if (!port) {
        let  port = new SerialPort(portOptions , () => {
          console.log('serial port opened');
        });
    //   }
      port.open(() => {
          port.on("data", (data: Buffer) => {
            console.log(data.toString());
          });
        });

      parentPort.postMessage({ type: 'SERIAL_PORT_STARTED' });
    }
  });
}

`

### What happens?

when i use seriaport in worker thread, it error:

`D:\codes\desktop-mmb>node dist/index.js
Received message from worker: { type: 'SERIAL_PORT_STARTED' }
FATAL ERROR: HandleScope::HandleScope Entering the V8 API without proper locking in place
 1: 00007FF6A8D7436F node::SetCppgcReference+15695
 2: 00007FF6A8CEC686 EVP_MD_meth_get_input_blocksize+76102
 3: 00007FF6A8CEE57C node::OnFatalError+252
 4: 00007FF6A974F549 v8::HandleScope::Initialize+137
 5: 00007FF6A97376FE v8::HandleScope::HandleScope+14
 6: 00007FF6A8D3FD8A napi_open_handle_scope+154
 7: 00007FFE7CD0AFEA 
 8: 00007FF6A8DD8A6B uv_update_time+491
 9: 00007FF6A8DD85E4 uv_run+900
10: 00007FF6A8DA94D2 node::SpinEventLoop+402 
11: 00007FF6A8C9CAB3 DH_get0_priv_key+151875 
12: 00007FF6A8D2E925 node::Start+3989        
13: 00007FF6A8D2D9C0 node::Start+48
14: 00007FF6A8B0D76C AES_cbc_encrypt+150940  
15: 00007FF6A9F3BCDC inflateValidate+17308   
16: 00007FFE9AF2257D BaseThreadInitThunk+29  
17: 00007FFE9B7CAA58 RtlUserThreadStart+40`

### What should have happened?

when i use it in mainthread, it works.
but i really need to use it in worker thread....

### Additional information

aaaa

jsharf, 2024-03-12:
Looks related to #1938.

egidiusmengelberg, 2024-09-04:
Yeah, having the same issue. I'm trying to integrate serialport in my iot nuxt app.

But nuxt is using workers so getting the same error

LouisCassany, 2024-10-17:
@egidiusmengelberg did you get any success? Trying to do exactly the same.

egidiusmengelberg, 2024-10-17:
I developed by constantly building the application and running it afterwards. Running the production build does not use workers which means serialport is working fine.

LouisCassany, 2024-10-17:
Seems like it's on the nitro roadmap [roadmap](https://github.com/nuxt/nuxt/discussions/28050).

Martin-Schlodinski, 2024-12-08:
same problem here...

LouisCassany, 2025-04-14:
> I'm also facing this issue when running serialport version 13 on node 22 on a RPi 4. The weird thing is that it's working completely fine on x64 ubuntu. What is the difference between the two?

Are you running both code in a worker thread? With nodeJS?

andrei-tatar, 2025-04-14:
@LouisCassany I managed to do a bit more testing. It seems that you can send data just fine, the problem is once you receive data.
Plugged in a dummy usb<->serial adapter, the code works fine. Once I bridge RX <-> TX so the data is echoed back, it fails immediately with the same error as on the raspberry pi.

LouisCassany, 2025-04-14:
Do you mind sharing a minimal working exemple? I just want to confirm I had the same behavior.

andrei-tatar, 2025-04-14:
Sure, this works for me just fine:

```
const { Worker } = require("node:worker_threads");

const worker = new Worker(`
    const { SerialPort } = require("serialport");

    const port = new SerialPort({
        baudRate: 115200,
        path: "/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0",
        autoOpen: true,
    });

    port.on("open", () => {
        console.log("Port opened");

        let i = 0;
        setInterval(() => {
            console.log("Sending data " + (i++));
            port.write("Hello from the worker thread!");
        }, 1000);

        // port.on("data", (data) => {
        //     console.log("Received data: " + data.toString());
        // });
    });

`, { eval: true });
```

As soon as I uncomment the receive part, it fails with:
```
FATAL ERROR: HandleScope::HandleScope Entering the V8 API without proper locking in place
----- Native stack trace -----

 1: 0xb80b7c node::OnFatalError(char const*, char const*) [node]
 2: 0xeeda66 v8::Utils::ReportApiFailure(char const*, char const*) [node]
 3: 0xeef12c v8::HandleScope::HandleScope(v8::Isolate*) [node]
 4: 0xc5b8fa napi_open_handle_scope [node]
 5: 0x7330e24190e7 Poller::onData(uv_poll_s*, int, int) [/home/andrei/proiecte/auto-mower/mower/node_modules/@serialport/bindings-cpp/prebuilds/linux-x64/@serialport+bindings-cpp.glibc.node]
 6: 0x18c3d4b  [node]
 7: 0x18afff7 uv_run [node]
 8: 0xbc7be6 node::SpinEventLoopInternal(node::Environment*) [node]
 9: 0xd0ae44  [node]
10: 0xd0b8dd node::NodeMainInstance::Run() [node]
11: 0xc6fc8f node::Start(int, char**) [node]
12: 0x73310982a1ca  [/lib/x86_64-linux-gnu/libc.so.6]
13: 0x73310982a28b __libc_start_main [/lib/x86_64-linux-gnu/libc.so.6]
14: 0xbc430e _start [node]
FATAL ERROR: v8::HandleScope::CreateHandle() Cannot create a handle without a HandleScope
----- Native stack trace -----

 1: 0xb80b7c node::OnFatalError(char const*, char const*) [node]
 2: 0xeeda66 v8::Utils::ReportApiFailure(char const*, char const*) [node]
 3: 0x1093c32 v8::internal::HandleScope::Extend(v8::internal::Isolate*) [node]
 4: 0xeef348 v8::EscapableHandleScope::EscapableHandleScope(v8::Isolate*) [node]
 5: 0xcba136 node::GetCurrentStackTrace(v8::Isolate*, int) [node]
 6: 0xc0d79b node::DumpJavaScriptBacktrace(_IO_FILE*) [node]
 7: 0xb80b88 node::OnFatalError(char const*, char const*) [node]
 8: 0xeeda66 v8::Utils::ReportApiFailure(char const*, char const*) [node]
 9: 0xeef12c v8::HandleScope::HandleScope(v8::Isolate*) [node]
10: 0xc5b8fa napi_open_handle_scope [node]
11: 0x7330e24190e7 Poller::onData(uv_poll_s*, int, int) [/home/andrei/proiecte/auto-mower/mower/node_modules/@serialport/bindings-cpp/prebuilds/linux-x64/@serialport+bindings-cpp.glibc.node]
12: 0x18c3d4b  [node]
13: 0x18afff7 uv_run [node]
14: 0xbc7be6 node::SpinEventLoopInternal(node::Environment*) [node]
15: 0xd0ae44  [node]
16: 0xd0b8dd node::NodeMainInstance::Run() [node]
17: 0xc6fc8f node::Start(int, char**) [node]
18: 0x73310982a1ca  [/lib/x86_64-linux-gnu/libc.so.6]
19: 0x73310982a28b __libc_start_main [/lib/x86_64-linux-gnu/libc.so.6]
20: 0xbc430e _start [node]
Aborted (core dumped)
```