serialport/node-serialport #2752: SerialPort has no construct signatures.

AdroitBit, 2024-03-11:
### SerialPort Version

12.0.0

### Node Version

v18.16.1

### Platform

Microsoft Windows NT 10.0.22622.0 x64

### Architecture

x64

### What steps will reproduce the bug?

Run this code 

// App.tsx
import SerialPort from 'serialport';
function App() {
  return (
    <>
      <p>fuck</p>
      <button onClick={function(){
        const port = new SerialPort({
          path: 'COM3',
          baudRate: 9600
        });
      }}>Test</button>
    </>
    
  );
}

export default App;

### What happens?

No button appeared and it seems there is the problem in browser's console.

The error in vscode says This expression is not constructable.
  Type 'typeof import("c:/Users/yanot/OneDrive/Desktop/small-CSI-Localization/csi_localization_full/node_modules/serialport/dist/index")' has no construct signatures.

The error in browser console says 
[image]

### What should have happened?

The button "Test" appeared and no error should happened probably.
but this code is to just to test serialport purely.

AdroitBit, 2024-03-12:
It seems to me. serialport is not usable at all?
console.log(serialPort); and the whole page is just dead

xiaodahai, 2024-12-16:
It seems to me. serialport is not usable at all?
console.log(serialPort); and the whole page is just dead

reconbot, 2024-12-24:
This can't work in a browser.

closed 2024-12-24 by reconbot