serialport/node-serialport #2773: After packing with electron, the running program reported an error that the module could not be found

CaoXinHao, 2024-04-08:
### SerialPort Version

12.0.0

### Node Version

16.14.1

### Electron Version

28.2.1

### What steps will reproduce the bug?

  
My project environment: pnpm + vite5 + electron-builder24.13.3

After packaging with electron-builder, running an error occurred:
[image]
  
I tried to install the missing module and found that all the serialport dependencies were missing

 

### What happens?

  
The packaged exe cannot be opened

 

### What should have happened?

  
Normal open

DaYao0704, 2024-04-19:
I had the same problem.Has the problem been solved now?

CaoXinHao, 2024-04-19:
I have a problem with pnpm, which is going to have a problem in electron.  I can change yarn 

> 我有同样的问题。现在问题解决了吗？

closed 2024-04-19 by CaoXinHao