serialport/node-serialport #2776: Hanging promise after close() in flowing mode

jichu4n, 2024-04-12:
### SerialPort Version

12.0.0

### Node Version

v18.18.2

### Platform

Linux devfox 6.8.2-arch2-1 #1 SMP PREEMPT_DYNAMIC Thu, 28 Mar 2024 17:06:35 +0000 x86_64 GNU/Linux

### Architecture

x64

### Hardware or chipset of serialport

Prolific Technology, Inc. PL2303 Serial Port / Mobile Phone Data Cable

### What steps will reproduce the bug?

1. Attach a USB-to-serial adapter (mine has a Prolific PL2303 chip) and leave it disconnected so no data is available to read.
2. Create a `SerialPort` instance and attach a `data` callback: `serialport.on('data', (data) => { ... })`.
3. Wait a few seconds, then close the `SerialPort` instance: `serialport.close(...)`. Adding `pause()` or `destroy()` is also fine.

Minimal repro: [serialport-test.ts](https://gist.github.com/jichu4n/b654ef675a0beb87ac3349b0f1babaf8#file-serialport-test-ts)

### What happens?

Even after calling `close()` or `destroy()`, the promise created on [unix-read.ts:41](https://github.com/serialport/bindings-cpp/blob/39fd4be0904434a0ec72ee88e04d54bfa1645667/lib/unix-read.ts#L41) never resolves / rejects. As a result, Node.js is prevented from exiting indefinitely.

See [sample output](https://gist.github.com/jichu4n/b654ef675a0beb87ac3349b0f1babaf8#file-output-txt) from the minimal repro program above.

### What should have happened?

The program should be able to exit cleanly after `close()` / `destroy()` is called.

rotu, 2024-06-24:
Possibly resolved by #2831

xty19, 2024-07-12:
SerialPort Version
12.0.0

Node Version
V18.16.1 (MACOS)  V20.15.0(WIN10,WIN11 )  

Electron Version
No response

Platform
MACOS12.1   WIN11 WIN10

Architecture
x64

Hardware or chipset of serialport
FT232R USB UART

I use serialport  and I use it in node-red  MACOS OS,and it runs good. But when It runs on WIN10 or WIN11 , it occurs some problems. ACCESS DENIED.  It took me 3days ,and finally I found .When I close the port ,even await ,It still cant close clearly.I have to add some delay to solve the problem.
So can anyone find the solution？