serialport/node-serialport #2798: spawn EINVAL

zN3utr4l, 2024-05-09:
### SerialPort Version

12.0.0

### Node Version

18.20.2

### Platform

Microsoft Windows NT 10.0.22631.0 x64

### Architecture

x64

### What steps will reproduce the bug?

$Env:npm_config_arch="arm" ; $Env:npm_config_platform="linux" ; yarn install --inline-builds

sqlite3 or canvas in package.json

### What happens?

```
node:internal/child_process:414
    throw errnoException(err, 'spawn');
    ^

Error: spawn EINVAL
    at ChildProcess.spawn (node:internal/child_process:414:11)
    at Object.spawn (node:child_process:761:9)
    at build (G:\Roboqbo\qbo-5\server\node_modules\@serialport\bindings-cpp\node_modules\node-gyp-build\bin.js:29:8)
    at preinstall (G:\Roboqbo\qbo-5\server\node_modules\@serialport\bindings-cpp\node_modules\node-gyp-build\bin.js:38:32)
    at G:\Roboqbo\qbo-5\server\node_modules\@serialport\bindings-cpp\node_modules\node-gyp-build\bin.js:11:7
    at ChildProcess.exithandler (node:child_process:430:5)
    at ChildProcess.emit (node:events:517:28)
    at maybeClose (node:internal/child_process:1098:16)
    at Socket.<anonymous> (node:internal/child_process:450:11)
    at Socket.emit (node:events:517:28) {
  errno: -4071,
  code: 'EINVAL',
  syscall: 'spawn'
}

Node.js v18.20.2
```

### What should have happened?

build

### Additional information

Look at this: https://github.com/nodejs/node/issues/52681

heiHenry, 2024-11-19:
> same problem here

Hello!Have you solved the problem?

yurih567, 2025-02-10:
PR #2962 fix this problem