serialport/node-serialport #2811: Please release missing native build for Arm64 armv=8?

ironsheep, 2024-05-24:
### SerialPort Version

12.0.0

### Node Version

18.18.2

### Electron Version

28.2.8

### Platform

platform=win32 arch=arm64 runtime=electron abi=119 uv=1 armv=8 libc=glibc

### Architecture

Arm64 armv=8

### Hardware or chipset of serialport

N/A

### What steps will reproduce the bug?

I have SerialPort built into my VSCode extension. It runs well on many platforms.
I have a Mac Studio - Apple M2 Ultra, 192GB running macOS Sonoma 14.5.
I'm running Windows 11 in Parallels Desktop for Max (Pro Edition) v19.4.0

I install the [spin2](https://marketplace.visualstudio.com/items?itemName=IronSheepProductionsLLC.spin2) extension into VSCode and I open a spin2 file.  The extension crashes with the error shown in "What happens" below.  It appears that one of the supporting binaries is not available for download: 
`No native build was found for platform=win32 arch=arm64 runtime=electron abi=119 uv=1 armv=8 libc=glibc node=18.18.2 electron=28.2.8`
 
How can we get this native build added to the distribution?

This is one of my primary development platforms (I am using Windows 10, Linux raspiOS, macOS Sonoma, and Windows 11 in Parallels) for my development of this extension.

Here is the unreleased version of the extension that exhibits the crash...
[spin2-2.2.19.vsix.zip](https://github.com/serialport/node-serialport/files/15427095/spin2-2.2.19.vsix.zip)

### What happens?

This is the crash output from the Spin2 Extension Host log

```
2024-05-23 21:02:53.368 [info] ExtensionService#_doActivateExtension IronSheepProductionsLLC.spin2, startup: false, activationEvent: 'onLanguage:spin2'
2024-05-23 21:02:53.552 [error] Activating extension IronSheepProductionsLLC.spin2 failed due to an error:
2024-05-23 21:02:53.552 [error] Error: No native build was found for platform=win32 arch=arm64 runtime=electron abi=119 uv=1 armv=8 libc=glibc node=18.18.2 electron=28.2.8
    loaded from: c:\Users\stephen\.vscode\extensions\ironsheepproductionsllc.spin2-2.2.19\client\node_modules\@serialport\bindings-cpp
    at Function.load.resolve.load.path (c:\Users\stephen\.vscode\extensions\ironsheepproductionsllc.spin2-2.2.19\client\node_modules\node-gyp-build\node-gyp-build.js:60:9)
    at load (c:\Users\stephen\.vscode\extensions\ironsheepproductionsllc.spin2-2.2.19\client\node_modules\node-gyp-build\node-gyp-build.js:22:30)
    at Object.<anonymous> (c:\Users\stephen\.vscode\extensions\ironsheepproductionsllc.spin2-2.2.19\client\node_modules\@serialport\bindings-cpp\dist\load-bindings.js:11:46)
    at Module.o._compile (c:\Users\stephen\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:4:1271)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1326:10)
    at Module.load (node:internal/modules/cjs/loader:1126:32)
...
```

### What should have happened?

It should locate the native build binary and continue to run as normal

### Additional information

If you need anything else you can reach me at stephen@ironsheep.biz

iamshiv4m, 2024-06-26:
@ironsheep , is this fixed? i am getting the same error

Komodood, 2024-07-06:
Same problem

ironsheep, 2024-07-07:
I've just found out tonight that this appears to be a packaging issue. The platform binaries are prebuilt. But the runtime loader needs to find them in a specific location.  I had to create  a receiving directory that my extension packages. I then has to copy the prebuilt .node files into this new location.

```bash
mkdir -p ./client/prebuilds && cp -pr ./client/node_modules/@serialport/bindings-cpp/prebuilds/* ./client/prebuilds/
```

My extension distributes both a `./client/out/extension.js` file and also a `./server/out/server.js' file. It now also distributes this new  `./client/prebuilds/` folder as well.  

Now this is all working for me.

ironsheep, 2024-07-07:
AUGH! This only solved part of my distribution problem in that **win32-arm64/** is NOT one of the prebuilt platform binaries. I need to run Windows 11 in Parallels Desktop on a Mac Sudio M2 Ultra-based machine.

The following are there tho'
```bash
android-arm/  
android-arm64/  
darwin-x64+arm64/  
linux-arm/  
linux-arm64/  
linux-x64/  
win32-ia32/  
win32-x64/
```

Komodood, 2024-07-07:
> AUGH! Это решило только часть моей проблемы с дистрибуцией, поскольку **win32-arm64/** НЕ является одним из готовых двоичных файлов платформы. Мне нужен **win32-arm64/** для запуска Windows 11 в Parallels Desktop на машине Mac Sudio M2 Ultra.
> 
> Однако есть и следующее:
> 
> ```shell
> android-arm/  
> android-arm64/  
> darwin-x64+arm64/  
> linux-arm/  
> linux-arm64/  
> linux-x64/  
> win32-ia32/  
> win32-x64/
> ```

so u mean, this liat of archs was builded?

ironsheep, 2024-07-07:
Yes, these are all found in `node_modules/@serialport/bindings-cpp/prebuilds`

ncls-alien, 2024-10-15:
Needs fix ASAP

HEEMO-95, 2025-05-26:
no support for ARM yet ?