serialport/node-serialport #2884: install error in win10

kerwinxu, 2024-08-22:
### SerialPort Version

12.0.0

### Node Version

v20.16.0

### Platform

Microsoft Windows NT 10.0.19045.0 x64

### Architecture

x64

### What steps will reproduce the bug?

npm install serialport --build-from-source 

### What happens?

npm warn cleanup Failed to remove some directories [
npm warn cleanup   [
npm warn cleanup     'C:\\Users\\kerwin\\Desktop\\2\\node_modules\\@serialport\\bindings-cpp',
npm warn cleanup     [Error: EPERM: operation not permitted, rmdir 'C:\Users\kerwin\Desktop\2\node_modules\@serialport\bindings-cpp\node_modules'] {
npm warn cleanup       errno: -4048,
npm warn cleanup       code: 'EPERM',
npm warn cleanup       syscall: 'rmdir',
npm warn cleanup       path: 'C:\\Users\\kerwin\\Desktop\\2\\node_modules\\@serialport\\bindings-cpp\\node_modules'
npm warn cleanup     }
npm warn cleanup   ]
npm warn cleanup ]
npm error code 1
npm error path C:\Users\kerwin\Desktop\2\node_modules\@serialport\bindings-cpp
npm error command failed
npm error command C:\WINDOWS\system32\cmd.exe /d /s /c node-gyp-build
npm error node:internal/child_process:421
npm error     throw new ErrnoException(err, 'spawn');
npm error     ^
npm error
npm error Error: spawn EINVAL
npm error     at ChildProcess.spawn (node:internal/child_process:421:11)
npm error     at Object.spawn (node:child_process:761:9)
npm error     at build (C:\Users\kerwin\Desktop\2\node_modules\node-gyp-build\bin.js:29:8)
npm error     at preinstall (C:\Users\kerwin\Desktop\2\node_modules\node-gyp-build\bin.js:38:32)
npm error     at Object.<anonymous> (C:\Users\kerwin\Desktop\2\node_modules\node-gyp-build\bin.js:15:3)
npm error     at Module._compile (node:internal/modules/cjs/loader:1358:14)
npm error     at Module._extensions..js (node:internal/modules/cjs/loader:1416:10)
npm error     at Module.load (node:internal/modules/cjs/loader:1208:32)
npm error     at Module._load (node:internal/modules/cjs/loader:1024:12)
npm error     at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:174:12) {
npm error   errno: -4071,
npm error   code: 'EINVAL',
npm error   syscall: 'spawn'
npm error }
npm error
npm error Node.js v20.16.0
npm error A complete log of this run can be found in: C:\Users\kerwin\AppData\Local\npm-cache\_logs\2024-08-22T13_23_00_136Z-debug-0.log

### What should have happened?

successfull

heiHenry, 2024-11-18:
have you resolved this problem?

kerwinxu, 2024-11-27:
> have you resolved this problem?

my resolve this problem .  
[](https://kerwinxu.github.io/posts/vue,electrom,serialport/)

closed 2024-11-27 by kerwinxu

kerwinxu, 2024-11-27:
> have you resolved this problem?

Yes .