serialport/node-serialport #2957: `NOMINMAX` missing to successfully rebuild for Electron `v33.2.0`

LeoLTM, 2024-12-16:
### SerialPort Version

v12.0.0

### Node Version

v20.18.1

### Electron Version

v33.2.0

### Platform

Microsoft Windows NT 10.0.19045.0 x64

### Architecture

x64

### What steps will reproduce the bug?

Setup an Vite Typescript Electron project, install serialport and Firmata using npm, run `npm run start` which runs

`electron-forge start` 

### What happens?

While preparing the application, the native dependencies are being rebuilt for the electron-internal Node.js version. This fails because there apparently are syntax errors in the "@serialport" dependency in Firmata.

The output is similar to [this issue in the `nan` project](https://github.com/nodejs/nan/issues/968).

I tried Node 18 - 22 and the problem occurred every single time.
The only way I was able to fix it was by manually adding the following to `node_modules\firmata\node_modules\@serialport\bindings\binding.gyp`

```json
      'defines': [
          'NOMINMAX'
      ],
```

so that it looks like this: 

```json
      ['OS=="win"',
        {
            'defines': [
                'NOMINMAX'
            ],
          'sources': [
            'src/serialport_win.cpp'
          ],
          'msvs_settings': {
            'VCCLCompilerTool': {
              'ExceptionHandling': '2',
              'DisableSpecificWarnings': [ '4530', '4506' ],
            }
          }
        }
      ]
```

### What should have happened?

The Firmata / SerialPort dependency should just rebuild for the Electron Node.js version like every other native dependency.

### Additional information

Another project fixed this in their `bindings.gyp` like this ([see here](https://github.com/deepnest-io/Deepnest/pull/180/commits/b11f3347c3417219335c771ec5d03ea22873736a#diff-0d6aa7470314ca774fd7e07f67b35ecd3045fe5cb3a0e5c99f48bae897905d32)), that's why I thought it might fix it for serialport as well.

xiejay97, 2024-12-31:
Same issue!

rotu, 2025-02-28:
Fixed by https://github.com/serialport/bindings-cpp/pull/210