serialport/node-serialport #2972: @serialport/list, @serialport/terminal requires 'package.json' to run

serg3295, 2024-12-26:
### SerialPort Version

13.0.0

### Node Version

v20.18.1

### Platform

Linux Lenovo 6.8.0-51-generic #52-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec  5 13:09:44 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### Architecture

x64

### Hardware or chipset of serialport

esp32, esp8266

### What steps will reproduce the bug?

npx @serialport/terminal
npx @serialport/list

### What happens?

<details>
  <summary>Crash log</summary>

  ```bash
❯ npx @serialport/terminal
node:fs:448
    return binding.readFileUtf8(path, stringToFlags(options.flag));
                   ^

Error: ENOENT: no such file or directory, open '../package.json'
    at readFileSync (node:fs:448:20)
    at Object.<anonymous> (/home/serg/Projects/js/begin/node_modules/.pnpm/@serialport+terminal@13.0.0/node_modules/@serialport/terminal/dist/index.js:13:59)
    at Module._compile (node:internal/modules/cjs/loader:1469:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1548:10)
    at Module.load (node:internal/modules/cjs/loader:1288:32)
    at Module._load (node:internal/modules/cjs/loader:1104:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:173:12)
    at node:internal/main/run_main_module:28:49 {
  errno: -2,
  code: 'ENOENT',
  syscall: 'open',
  path: '../package.json'
}

Node.js v20.18.1

❯ npx @serialport/list
node:fs:448
    return binding.readFileUtf8(path, stringToFlags(options.flag));
                   ^

Error: ENOENT: no such file or directory, open '../package.json'
    at readFileSync (node:fs:448:20)
    at Object.<anonymous> (/home/serg/Projects/js/begin/node_modules/.pnpm/@serialport+list@13.0.0/node_modules/@serialport/list/dist/index.js:7:59)
    at Module._compile (node:internal/modules/cjs/loader:1469:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1548:10)
    at Module.load (node:internal/modules/cjs/loader:1288:32)
    at Module._load (node:internal/modules/cjs/loader:1104:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:173:12)
    at node:internal/main/run_main_module:28:49 {
  errno: -2,
  code: 'ENOENT',
  syscall: 'open',
  path: '../package.json'
}

❯ npx @serialport/repl
No arduinos found. You must specify a port to load.

For example:
        serialport-repl COM3
        serialport-repl /dev/tty.my-serialport

Node.js v20.18.1

```
  
</details>

### What should have happened?

The @serialport/list, @serialport/terminal  should run without additional 'package.json'

### Additional information

I added package.json:
```json
{
  "version": "0.1.0",
  "description": "testing"
}
```
the @serialport/list, @serialport/terminal started working.

ebaauw, 2025-01-09:
Problem is that `package.json` is now read using `readFileSync()`, which uses the current directory as base,  instead of `require()`, which uses the programme directory.

matt-oakes, 2025-05-12:
I have opened a small PR which fixes this issue https://github.com/serialport/node-serialport/pull/3034

In the meantime, you can edit the file referenced in the crash log and add `__dirname` to the path so it will run correctly.

tkurki, 2025-05-20:
I just bumped against this using `npx @serialport/list` per documentation at https://serialport.io/docs/bin-list.

awmartin, 2025-06-25:
Running into this issue as well. @npx @serialport/list shows this error.

When `package.json` is in the parent folder, I see this error:

```shell
8674 ± : npx @serialport/list                                                                                                                                                                                   [-29171663m] ✖ ✹ ✭
/Users/awmartin/.npm/_npx/b96e195ab65bff9d/node_modules/@serialport/list/dist/index.js:9
commander_1.program.version(version).description('List available serial ports').addOption(formatOption).parse(process.argv);
                                    ^

TypeError: Cannot read properties of undefined (reading 'description')
    at Object.<anonymous> (/Users/awmartin/.npm/_npx/b96e195ab65bff9d/node_modules/@serialport/list/dist/index.js:9:37)
    at Module._compile (node:internal/modules/cjs/loader:1241:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1295:10)
    at Module.load (node:internal/modules/cjs/loader:1091:32)
    at Module._load (node:internal/modules/cjs/loader:938:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:83:12)
    at node:internal/main/run_main_module:23:47
```